stdr-simulator-ros-pkg/stdr_simulator #171: Bug with footprint loading

etsardou, 2015-08-22:
When a robot is loaded (either added in the map or loaded in the robot creator) the footprint is uninitialized. Probably a bug in the parser.

etsardou, 2015-08-22:
Actually many elements are not loaded correctly.

Notes:
- Affects both xml and yaml files
- OS: Ubuntu 14.04, ROS Indigo
- yaml-cpp: 0.5.1

- The robot_msgs is wrong in the parser msg creation
- stdr_yaml_parser seems to create the correct Node* structure with the correct values

etsardou, 2015-08-22:
Check the code in the ```kinematic_models_support``` branch. To reproduce:

- launch the ```roslaunch stdr_launchers server_with_map_and_gui_plus_robot.launch```
- The predefined robot is spawned correctly. ROS_ERROR messages are evident in the console. These exist here: https://github.com/stdr-simulator-ros-pkg/stdr_simulator/blob/motion_controllers_support/stdr_parser/src/stdr_parser_msg_creator.cpp#L168
- Try to add a new predefined robot (e.g. pandora.yaml which is the one loaded in the previous launcher). The string in the yaml and the Node* structure is correct but the ```atof``` seems to fail.

oscar-lima, 2015-09-28:
when launching :

roslaunch stdr_launchers server_with_map_and_gui_plus_robot.launch

in branch : motion_controllers_support

i get this errors in console:

[ERROR] [/robot_spawn_oscar_MacBook_11898_4164890650817676037]: Footprint_specifications found
[ERROR] [/robot_spawn_oscar_MacBook_11898_4164890650817676037]: The value is: 0.200000002980232
[ERROR] [/robot_spawn_oscar_MacBook_11898_4164890650817676037]: 0.2
[ERROR] [/robot_spawn_oscar_MacBook_11898_4164890650817676037]: The radius is 0.200000

but the robot seems to be working fine.

etsardou, 2015-09-29:
Ok great, thank you :)

commit d90a8eb3 (2015-09-29): fixes #171
closed 2015-10-02 by etsardou