stdr-simulator-ros-pkg/stdr_simulator #175: robot spawn crashes when loading yaml omni robot

oscar-lima, 2015-09-27:
it is able to load xml robots but when loading yaml robot in the same format gives this error:

etsardou, 2015-09-28:
Hi Oscar, I believe you have misspelled "specifications". Please check again with the proper tag ```kinematic_specifications```.

etsardou, 2015-09-28:
And just a question: Does that bug occurs to you too? https://github.com/stdr-simulator-ros-pkg/stdr_simulator/issues/171

Thnx

oscar-lima, 2015-09-28:
Your are correct, it was a miss spelling error, i commented on issue 171 already. This issue can be then closed now

oscar-lima, 2015-09-28:
just as a comment, when launching the yaml robot in:

https://github.com/oscar-lima/cob3_navigation_sim/blob/hydro/ros/config/cob3-1.yaml

I get this errors:

[ERROR] [/robot_spawn_oscar_MacBook_12590_2934590099813672843]: Footprint_specifications found
[ERROR] [/robot_spawn_oscar_MacBook_12590_2934590099813672843]: The value is: 0.5
[ERROR] [/robot_spawn_oscar_MacBook_12590_2934590099813672843]: 0.5
[ERROR] [/robot_spawn_oscar_MacBook_12590_2934590099813672843]: The radius is 0.500000

although the robot works fine.

I guess this was your question regarding issue 171.

closed 2015-09-28 by oscar-lima