stdr-simulator-ros-pkg/stdr_simulator #187: How to understand the obstacle avoidance function?

libing64, 2016-04-06:
The obstacle avoidance method in stdr_sample seems very simple, is there any paper talking about this method?
  
void ObstacleAvoidance::callback(const sensor_msgs::LaserScan& msg)
  {
    scan_ = msg;
    float linear = 0, rotational = 0;
    for(unsigned int i = 0 ; i < scan_.ranges.size() ; i++)
    {
      float real_dist = scan_.ranges[i];
      linear -= cos(scan_.angle_min + i * scan_.angle_increment) 
        / (1.0 + real_dist * real_dist);
      rotational -= sin(scan_.angle_min + i * scan_.angle_increment) 
        / (1.0 + real_dist * real_dist);
    }
    geometry_msgs::Twist cmd;
    
    linear /= scan_.ranges.size();
    rotational /= scan_.ranges.size();
    
    //~ ROS_ERROR("%f %f",linear,rotational);
    
    if(linear > 0.3)
    {
      linear = 0.3;
    }
    else if(linear < -0.3)
    {
      linear = -0.3;
    }

    cmd.linear.x = 0.3 + linear;
    cmd.angular.z = rotational;
    cmd_vel_pub_.publish(cmd);
  }
}

etsardou, 2016-04-06:
Hey, it seems simple because it is simple! This was created in order to showcase how you can use the topics of STDR to do something autonomous, and not to actually perform robust obstacle avoidance. The method followed tries to produce a linear and a rotational velocity according to these rules:

- The linear velocity is constant (0.3 m/sec) and a corrective coefficient is added, taking into consideration mostly the laser rays in the front side of the robot (where the cosine takes its higher value)
- The rotational velocity takes values according to the sinus value, i.e. it is affected more by the side rays. Since sine produces a different sign for each side, it can be directly used as the velocity.

In both cases, the rays measurements are taken into consideration, so as for the closer obstacles to have a larger effect in the avoidance.

closed 2016-04-06 by etsardou

libing64, 2016-04-06:
@etsardou Great thanks.