stdr-simulator-ros-pkg/stdr_simulator #193: ROS Kinetic Release

rohbotics, 2016-07-11:
stdr_simulator is blocking turtlebot_simulator, it would be great if you can get it released on kinetic (See https://github.com/turtlebot/turtlebot/issues/219) 

I did a quick test, and stdr_simulator compiles on Ubuntu Xenial with Kinetic.

czalidis, 2016-07-11:
Unfortunately we are long due for a release on all distros (Indigo, Jade and Kinetic). We will try to push a release soon, after merging some pending issues! If this takes a lot of time we might release without new changes.

tfoote, 2016-07-11:
A quick release would be great even if it's just the old version so we can get downstream packages released. aka turtlebot_simulator. Otherwise we may consider dropping stdr_simulator as a dependency until it's released just to make progress.

czalidis, 2016-07-11:
Ok then, I will consider pushing a release of the old version, sometime this week.

czalidis, 2016-07-18:
Pushed a release of a new version (with minor changes) for Kinetic.

closed 2016-07-18 by czalidis