stdr-simulator-ros-pkg/stdr_simulator #195: STDR robot spawn - Error   -   Unable to load or create robot in stdr_simulator

Oskedosk, 2016-08-22:
The problem occurs when I do the following:
1. roscd stdr_resources
2. roslaunch stdr_launchers server_with_map_and_gui.launch
3. load or create a robot using the buttons in the GUI

Running roslaunch stdr_launchers server_with_map_and_gui_plus_robot.launch gives the same error.

Doing the above three points gives the following error message in the GUI:
**STDR robot spawn - Error
Could not spawn robot. Robot didn't respond within time or registered with incorrect name.**

The output form the Terminal is:

> 
> started roslaunch server http://192.168.13.28:50243/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: indigo
>  * /rosversion: 1.11.20
> 
> NODES
>   /
>     robot_manager (nodelet/nodelet)
>     stdr_gui_node_oskar_HP_17864_4557666770542936268 (stdr_gui/stdr_gui_node)
>     stdr_server (stdr_server/stdr_server_node)
>     world2map (tf/static_transform_publisher)
> 
> ROS_MASTER_URI=http://192.168.13.28:11311
> 
> core service [/rosout] found
> process[robot_manager-1]: started with pid [17882]
> process[stdr_server-2]: started with pid [17883]
> process[world2map-3]: started with pid [17884]
> process[stdr_gui_node_oskar_HP_17864_4557666770542936268-4]: started with pid [17885]
> [ INFO] [1471873548.690764052]: Loading map from image "/opt/ros/indigo/share/stdr_resources/maps/sparse_obstacles.png"
> [ INFO] [1471873548.713014357]: Read a 775 X 746 map @ 0.020 m/cell
> [ERROR] [1471873556.975768095]: Failed to load nodelet [robot0] of type [stdr_robot/Robot] even after refreshing the cache: Failed to load library /opt/ros/indigo/lib/libstdr_robot_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libstdr_omni_motion_controller.so: cannot open shared object file: No such file or directory)
> [ERROR] [1471873556.975830483]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib/libstdr_robot_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libstdr_omni_motion_controller.so: cannot open shared object file: No such file or directory)

Oskedosk, 2016-08-22:
I'm running Ubuntu 14.04.5 LTS

Dulluhan, 2016-08-27:
I am having a similar issue with the tutorial provided launch files. It seems that the gui can handle up to loading a server and a map, but when I try to load a robot, it crashes. The whole gui ends up being disabled till it got SIGKILL. 

I've tried loading a premade robot and creating my own in the gui. It crashes the exact moment I left click to place down the robot.

czalidis, 2016-08-28:
Yep, this is a bug, the `omni_motion_controller` library is missing from the install targets. Thanks for catching this!

The latest commit should resolve this issue, although you have to wait until the next binary release to get the latest changes.

Compiling from source should work fine, even for the old version, the issue only occurs when installing form apt packages.

closed 2016-08-28 by czalidis

Oskedosk, 2016-09-01:
Works perfectly now, thanks!

CoderSquirrel, 2016-09-02:
"Compiling from source should work fine, even for the old version, the issue only occurs when installing form apt packages." 
sorry i'm noob on this, but once i cloned howi compile?

etsardou, 2016-09-03:
You must first [create a catkin workspace](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment#Create_a_ROS_Workspace) and then [clone and build STDR in it](http://wiki.ros.org/stdr_simulator/Tutorials/Set%20up%20STDR%20Simulator#Get_STDR_Simulator_from_Github) ;)

ex1t-vtc, 2016-10-25:
i have the same problem here... tried both installing from apt packages and compiling from source. Only difference is that i am running ROS kinetic. 

I checked if the target stdr_omni_motion_controller is missing but it is in the code.

----------------------------------------------------------------------------------------------------------------------------------

server_no_map.launch
server_with_map_and_gui.launch
server_with_map_and_gui_plus_robot.launch
server_with_map.launch
stribor@stribor-ThinkPad-L540:~$ roslaunch stdr_launchers server_with_map_and_gui_plus_robot.launch 
... logging to /home/stribor/.ros/log/6059c500-9aad-11e6-9486-e8b1fcef2dab/roslaunch-stribor-ThinkPad-L540-4322.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://stribor-ThinkPad-L540:33102/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.5

NODES
  /
    robot_manager (nodelet/nodelet)
    robot_spawn_stribor_ThinkPad_L540_4322_7858200342873178241 (stdr_robot/robot_handler)
    stdr_gui_node_stribor_ThinkPad_L540_4322_5353934228803564853 (stdr_gui/stdr_gui_node)
    stdr_server (stdr_server/stdr_server_node)
    world2map (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [4335]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6059c500-9aad-11e6-9486-e8b1fcef2dab
process[rosout-1]: started with pid [4348]
started core service [/rosout]
process[robot_manager-2]: started with pid [4351]
process[stdr_server-3]: started with pid [4352]
process[world2map-4]: started with pid [4353]
process[stdr_gui_node_stribor_ThinkPad_L540_4322_5353934228803564853-5]: started with pid [4354]
process[robot_spawn_stribor_ThinkPad_L540_4322_7858200342873178241-6]: started with pid [4355]
[ INFO] [1477398012.279087315]: Loading map from image "/opt/ros/kinetic/share/stdr_resources/maps/sparse_obstacles.png"
[ INFO] [1477398012.292859504]: Read a 775 X 746 map @ 0.020 m/cell
[ERROR] [1477398012.648202594]: Failed to load nodelet [robot0] of type [stdr_robot/Robot] even after refreshing the cache: Failed to load library /opt/ros/kinetic/lib/libstdr_robot_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libstdr_omni_motion_controller.so: cannot open shared object file: No such file or directory)
[ERROR] [1477398012.648316096]: The error before refreshing the cache was: Failed to load library /opt/ros/kinetic/lib/libstdr_robot_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libstdr_omni_motion_controller.so: cannot open shared object file: No such file or directory)
terminate called after throwing an instance of 'stdr_robot::DoubleFrameIdException'
  what():  Could not spawn robot. Robot didn't respond within time or registered with incorrect name.
[robot_spawn_stribor_ThinkPad_L540_4322_7858200342873178241-6] process has died [pid 4355, exit code -6, cmd /opt/ros/kinetic/lib/stdr_robot/robot_handler add /opt/ros/kinetic/share/stdr_resources/resources/robots/pandora_robot.yaml 1 2 0 __name:=robot_spawn_stribor_ThinkPad_L540_4322_7858200342873178241 __log:=/home/stribor/.ros/log/6059c500-9aad-11e6-9486-e8b1fcef2dab/robot_spawn_stribor_ThinkPad_L540_4322_7858200342873178241-6.log].
log file: /home/stribor/.ros/log/6059c500-9aad-11e6-9486-e8b1fcef2dab/robot_spawn_stribor_ThinkPad_L540_4322_7858200342873178241-6*.log

czalidis, 2016-10-26:
The bug remains for the binary version (available through apt packages) since it's not yet updated with the new, fixed version of STDR Simulator. I will try to push a new release before the next sync.

In the meantime, compiling from source should work just fine. In your case, based on the attached output log, although you are supposed to have tried to compile STDR, you still using the binary version. That's why you get `Error string: Could not load library (Poco exception = libstdr_omni_motion_controller.so`. Two possible solutions:
- Before you launch STDR Simulation (the compiled version), you have to source the appropriate `devel/setup.bash` file. That will overwrite the binary version that you have already installed, from apt packages.
- Simply make sure that you have removed the binary packages, so you don't get confused. You can remove that by running something like `sudo apt-get remove ros-kinetic-stdr-*`. You still need to source the correct `devel/setup.bash` file thought.

ex1t-vtc, 2016-10-26:
it works now. tnx.

Scalpel78, 2016-12-11:
Any idea when this fix will be in the binary version? I'm experiencing the same issue.

wsnewman, 2017-01-19:
I'm trying to use this for a class.  We're using the work-around of compiling from source, but it would be much easier if the bug fix were propagated to the binary installation (we're using Indigo).
My thanks for your help.

MahmoudElShazly, 2017-01-25:
For those who want to have the fix as if it is in the installed version:

1 - Compile it from source
2 - Copy over the compiled files to the install location
cp -a {workspace}/devel/lib/.     /opt/ros/indigo/lib/

wsnewman, 2017-01-25:
thanks!  We did use the "work-around" of compiling from source.  This will
get us by for now.

It would be nice if users could do a normal sudo apt-get and have things
work, without these additional steps.  But at least there is a work-around.

Thanks again

czalidis, 2017-02-06:
I pushed a new release for all current ROS distros (Indigo, Jade, Kinetic). When the next sync takes place, the fixed version will be available through apt packages. I'll follow up when this happens.

czalidis, 2017-02-28:
It looks like the new version is now available for all current distros, so you should be able to use the simulator without any problems through apt packages. Let me know if there are any issues.

pitosalas, 2018-02-10:
I am still having an issue:

$ roscore
$ roslaunch stdr_launchers rviz.launch
$ roslaunch stdr_launchers server_no_map.launch // not sure if this is needed. Was trying things
$ rosrun stdr_robot robot_handler add resources/robots/pandora_robot.xml

...[ WARN] [1518269261.924496791]: Could not find stdr_server/spawn_robot action.

(This is from a clean install on Ubuntu 16.04, kinetic, running under Vmware, Mac.)

czalidis, 2018-02-12:
Hi @pitosalas, you're issue is different. What you're experiencing is the expected behavior. The simulator only allows spawning of robots only if a map is already loaded. You need to load a map before trying to spawn a robot, for example by modifying your 3rd command: `roslaunch stdr_launchers server_with_map.launch`.

commit 2c3ad991 (2016-08-28): Add forgotten install target for omni_motion_controller

Closes #195