stdr-simulator-ros-pkg/stdr_simulator #196: A "Hello world" type of tutorial is needed for beginners

Scalpel78, 2016-12-11:
Hi, I'm trying out STDR for the first time (and I'm a beginner to ROS as well).

I've been able to install, and get it running (had to use GitHub approach described on the tutorial "STDR simulator setup" because the version in apt-get is buggy).

When looking through the other tutorials I'm not really understanding what to do. I can "load a robot", but have no idea how to get it to move.

A walkthrough type of tutorial is really needed to get beginners, like myself, onboard.

Player/Stage has a really good tutorial at (http://player-stage-manual.readthedocs.io/). Unfortunately I couldn't get their software working, and that's why I've decided to try out STDR instead.

Basically - how can I get robot moving?

etsardou, 2016-12-12:
Hi Scalpel! Since STDR Simulator is entirely ROS based, you can move the robot by giving velocities in a ROS [Twist](http://docs.ros.org/api/geometry_msgs/html/msg/Twist.html) topic. The default topic for the first robot you spawn is ```robot0/cmd_vel```. 

You can teleoperate the robot [like this](http://wiki.ros.org/stdr_simulator/Tutorials/Teleop%20with%20teleop_twist_keyboard) or you can make ti autonomous [like this](https://github.com/stdr-simulator-ros-pkg/stdr_simulator/tree/indigo-devel/stdr_samples), where a simple obstacle avoidance is described.

In general check out tutorials [here](http://wiki.ros.org/stdr_simulator/Tutorials) and if you have some time, we would be glad if you could add some more, based on your experience.

If you're covered please close this issue, else leave it open till a more simple tutorial is created :)

Scalpel78, 2016-12-13:
Thanks, that helps a lot :) I'll close this issue, and try out the samples.

closed 2016-12-13 by Scalpel78